ofTheo/ofxKinectV2 #8: can't open camera "anymore"

huggenberg, 2015-03-09:
Thanks for all your work.
After following your notes and updating the firmware of the camera I finally could make the program run the other day. But for some reason it doesn't work anymore, tried disconnecting, restarting, reinstalling, everything. No infrared lights or any other indicators on the camera itself, only the white light on the adapter. I'm using a new model (1520). And the osx system report tells me that the NUISensor is connected using usb 3.

Any ideas?

Using libusbx v1.0.17.10852

Trying to open device 045E:02D8...
Trying to open device 045E:02C4...
libusbx: error [darwin_reset_device] ResetDevice: unknown error
ofProtonect::openKinect  Failed. - handle is NULL

fraco, 2015-06-13:
Hi huggenberg,

I just stuble upon this same issue. It was working the first time, after trying to run it again, it began showing this same error in the Debug console. Now, even though it runs, no image is shown or any sign that the Kinect is actually powering. If you had any way workaround I'd greatly appreciate if you could share your fix or any insights.

best,
fraco

[ error ] ofXml: loadFromBuffer(): DOM ERROR
[warning] ofXml: setTo(): empty document
Using libusbx v1.0.17.10852

Trying to open device 045E:02D8...
libusbx: error [darwin_reset_device] ResetDevice: unknown error
Trying to open device 045E:02C4...
ofProtonect::openKinect  Failed. - handle is NULL

fraco, 2015-06-13:
Here is the short answer answer: after having everything connected, unplug power connection and replug.

The long answer:
[issue1](https://github.com/ofTheo/ofxKinectV2/issues/1)

huggenberg, 2015-06-14:
Hey fraco

I didn't solve the problem until today because I realised that I need a
higher resolution for my project anyway. I'm now working with a 5MP
standard infrared camera.
As far as I remember I repluged the power connection as well and it still
didn't work. But will try again.

closed 2015-11-16 by ofTheo