ofTheo/ofxKinectV2 #10: Beginner Questions Reguarding documentation

SebastianVarma, 2015-06-15:
@ofTheo Hi.  Does this have the same documentation as libfreenect2?  Does it have the same documentation as ofxKinect?  Are there any diferences between its documentation and the ofxKinect documentation?  Where can I find the documentation?

ofTheo, 2015-06-16:
hi - documentation currently doesn't exist.  The main info is in the ReadMe file at https://github.com/ofTheo/ofxKinectV2/blob/master/README.md

libfreenect2 is much newer currently but the GPU decoding they are using doesn't play nice with OF. 
So I will wait to update the ofxKinect2 version until then.

ofTheo, 2015-06-16:
ps: let me know if you have any questions regarding the addon :)

closed 2015-06-17 by SebastianVarma