ofTheo/ofxKinectV2 #15: I can't build success。。。

rongloveyan, 2015-11-25:
[image]

mattfelsen, 2015-11-25:
I believe this addon has only been developed/tested on OS X. On Windows, you'll need to install an OpenCL SDK, possibly need to install libusb/libusbK, and maybe tweak some other stuff. There's some documentation in the libfreenect2 docs inside the repo under the `Installation > Window / Visual Studio` section. I'm not sure how much of it is relevant (I am fairly certain you won't have to bother with glfw, for example) but it's a good place to start!

https://github.com/ofTheo/ofxKinectV2/tree/master/libs/libfreenect2

ofTheo, 2015-11-30:
For Windows I would recommend this addon: https://github.com/elliotwoods/ofxKinectForWindows2
Its really good and uses the official Kinect SDK.

My addon is really more for linux/osx users who don't have access to an official SDK

closed 2015-11-30 by ofTheo