ofTheo/ofxKinectV2 #16: opencl_depth_packet_processor.cpp Compile Errors

brannondorsey, 2015-12-03:
Hi there, I'm trying to compile the example but am running into errors when compiling the `opencl_depth_packet_processor.cpp` file. __Note__: I am using OF v0.8.4, while the README says optimized for v0.9.0, could this be related?

```
/Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:73:44: warning: missing terminating '"' character [-Winvalid-pp-token]
static const std::string clInlineSource = R"(
                                           ^
/Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:73:43: error: use of undeclared identifier 'R'
static const std::string clInlineSource = R"(
                                          ^
/Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:455:2: warning: missing terminating '"' character [-Winvalid-pp-token]
)";
 ^
/Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:998:13: error: allocation of incomplete type 'libfreenect2::OpenCLDepthPacketProcessorImpl'
  impl_(new OpenCLDepthPacketProcessorImpl(deviceId))
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:29:
../../../addons/ofxKinectV2/libs/libfreenect2/include/libfreenect2/depth_packet_processor.h:168:7: note: forward declaration of 'libfreenect2::OpenCLDepthPacketProcessorImpl'
class OpenCLDepthPacketProcessorImpl;
      ^
/Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:1004:3: warning: deleting pointer to incomplete type 'libfreenect2::OpenCLDepthPacketProcessorImpl' may cause undefined behavior [-Wdelete-incomplete]
  delete impl_;
  ^      ~~~~~
In file included from /Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:29:
../../../addons/ofxKinectV2/libs/libfreenect2/include/libfreenect2/depth_packet_processor.h:168:7: note: forward declaration of 'libfreenect2::OpenCLDepthPacketProcessorImpl'
class OpenCLDepthPacketProcessorImpl;
      ^
/Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:1011:13: error: member access into incomplete type 'libfreenect2::OpenCLDepthPacketProcessorImpl'
  if ( impl_->config.MaxDepth != config.MaxDepth 
            ^
In file included from /Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:29:
../../../addons/ofxKinectV2/libs/libfreenect2/include/libfreenect2/depth_packet_processor.h:168:7: note: forward declaration of 'libfreenect2::OpenCLDepthPacketProcessorImpl'
class OpenCLDepthPacketProcessorImpl;
      ^
/Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:1015:10: error: member access into incomplete type 'libfreenect2::OpenCLDepthPacketProcessorImpl'
    impl_->programBuilt = false;
         ^
In file included from /Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:29:
../../../addons/ofxKinectV2/libs/libfreenect2/include/libfreenect2/depth_packet_processor.h:168:7: note: forward declaration of 'libfreenect2::OpenCLDepthPacketProcessorImpl'
class OpenCLDepthPacketProcessorImpl;
      ^
/Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:1016:10: error: member access into incomplete type 'libfreenect2::OpenCLDepthPacketProcessorImpl'
    impl_->programInitialized = false;
         ^
In file included from /Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:29:
../../../addons/ofxKinectV2/libs/libfreenect2/include/libfreenect2/depth_packet_processor.h:168:7: note: forward declaration of 'libfreenect2::OpenCLDepthPacketProcessorImpl'
class OpenCLDepthPacketProcessorImpl;
      ^
/Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:1018:17: error: member access into incomplete type 'libfreenect2::OpenCLDepthPacketProcessorImpl'
  else if (impl_->config.EnableBilateralFilter != config.EnableBilateralFilter
                ^
In file included from /Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:29:
../../../addons/ofxKinectV2/libs/libfreenect2/include/libfreenect2/depth_packet_processor.h:168:7: note: forward declaration of 'libfreenect2::OpenCLDepthPacketProcessorImpl'
class OpenCLDepthPacketProcessorImpl;
      ^
/Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:1022:10: error: member access into incomplete type 'libfreenect2::OpenCLDepthPacketProcessorImpl'
    impl_->programInitialized = false;
         ^
In file included from /Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:29:
../../../addons/ofxKinectV2/libs/libfreenect2/include/libfreenect2/depth_packet_processor.h:168:7: note: forward declaration of 'libfreenect2::OpenCLDepthPacketProcessorImpl'
class OpenCLDepthPacketProcessorImpl;
      ^
/Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:1025:8: error: member access into incomplete type 'libfreenect2::OpenCLDepthPacketProcessorImpl'
  impl_->config = config;
       ^
In file included from /Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:29:
../../../addons/ofxKinectV2/libs/libfreenect2/include/libfreenect2/depth_packet_processor.h:168:7: note: forward declaration of 'libfreenect2::OpenCLDepthPacketProcessorImpl'
class OpenCLDepthPacketProcessorImpl;
      ^
/Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:1026:13: error: member access into incomplete type 'libfreenect2::OpenCLDepthPacketProcessorImpl'
  if (!impl_->programBuilt)
[... 54 lines cut ...]
/Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:1066:8: error: member access into incomplete type 'libfreenect2::OpenCLDepthPacketProcessorImpl'
  impl_->ir_frame->timestamp = packet.timestamp;
       ^
In file included from /Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:29:
../../../addons/ofxKinectV2/libs/libfreenect2/include/libfreenect2/depth_packet_processor.h:168:7: note: forward declaration of 'libfreenect2::OpenCLDepthPacketProcessorImpl'
class OpenCLDepthPacketProcessorImpl;
      ^
/Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:1067:8: error: member access into incomplete type 'libfreenect2::OpenCLDepthPacketProcessorImpl'
  impl_->depth_frame->timestamp = packet.timestamp;
       ^
In file included from /Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:29:
../../../addons/ofxKinectV2/libs/libfreenect2/include/libfreenect2/depth_packet_processor.h:168:7: note: forward declaration of 'libfreenect2::OpenCLDepthPacketProcessorImpl'
class OpenCLDepthPacketProcessorImpl;
      ^
/Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:1068:8: error: member access into incomplete type 'libfreenect2::OpenCLDepthPacketProcessorImpl'
  impl_->ir_frame->sequence = packet.sequence;
       ^
In file included from /Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/src/opencl_depth_packet_processor.cpp:29:
../../../addons/ofxKinectV2/libs/libfreenect2/include/libfreenect2/depth_packet_processor.h:168:7: note: forward declaration of 'libfreenect2::OpenCLDepthPacketProcessorImpl'
class OpenCLDepthPacketProcessorImpl;
      ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
3 warnings and 20 errors generated.

```

ofTheo, 2015-12-03:
yes - that string literal will only work in 0.9.0 and up as its new in C++11 
sorry! it was the easiest way to include the CL code so it didn�t need to be loaded from a file.

brannondorsey, 2015-12-03:
Ok great I will try with of 0.9.0! Thanks for the quick reply. In the meantime I attempted to build with the Makefile system and got errors including `<thread>` inside `libfrenect2`:

```
Users/bdorse/Documents/of_v0.8.4_osx_release/addons/ofxKinectV2/libs/libfreenect2/include/internal/libfreenect2/threading.h:36:10: fatal error: 'thread' file not found
#include <thread>
```

Seems unrelated, but maybe worth mentioning.

brannondorsey, 2015-12-03:
Works great in v0.9.0 via XCode and that `<thread>` issue is gone as well. I've created a PR [here](https://github.com/ofTheo/ofxKinectV2/pull/17) adding `ofxGui` to `example/addons.make` :wink:

closed 2015-12-03 by brannondorsey