ofTheo/ofxKinectV2 #19: Can't allocate an array of constant size 0 in libfreenect2

BaptisteTheoriz, 2016-02-11:
Hi, 

I encounter an error when compiling a project with ofxKinectV2 on Windows 10, Visual Studio 2015 and of 0.9.2.

The line 48 of resource.cpp file in libfreenect2 lib cause a "cannot allocate an array of constant size 0" error (C2466) :

```
#ifdef RESOURCES_INC
#include "resources.inc.h"
#else
static ResourceDescriptor resource_descriptors[] = {};
static int resource_descriptors_length = 0;
#endif
```

It seems to be related to [this issue](https://github.com/OpenKinect/libfreenect2/issues/443) in libfreenect2, but it doesn't cause error when compiling libfreenect2 project, because RESOURCES_INC macro is defined so this code is not compiled (and so the error isn't really fixed)

I didn't see any "resources.inc.h" file in the addon, so I don't know if it could be possible to bypass this error, or if I should report this issue to libfreenect team ?

ofTheo, 2016-02-11:
Let me know if this works for you and I can fix it or you could send a PR.

```
static ResourceDescriptor * resource_descriptors;
static int resource_descriptors_length = 0;
```

Cheers!
Theo

BaptisteTheoriz, 2016-02-11:
Seems to work, at least it compiles and works for me.
Thanks !

closed 2016-02-11 by BaptisteTheoriz