ofTheo/ofxKinectV2 #23: Problem with libusb in windows

caelshadowhunter, 2016-06-30:
Hi,
I'm currently attempting to build in windows 10 and visual studio 2015. I would like to work with multiple KinectV2s, so using an extension which relies on the Microsoft SDK isn't an option for me. 

To fix a previous error, I have already used the fix suggested [here](https://github.com/ofTheo/ofxKinectV2/issues/19), but I'm now getting a compile error :
> invalid or corrupt file

with libusb-1.0.dll being the problematic file. 

Any idea how I can fix this one?

urarakajun, 2017-05-30:
Hi,
I also had the same problem.
I tried this but I could not solve it　↓
https://github.com/ofTheo/ofxKinectV2/pull/21

Did everyone solve it?

windows10 + vs2017 + 0.9.8

a172862967, 2017-10-21:
Anyone solve it?

a172862967, 2017-10-23:
@ofTheo Plz Help

ofTheo, 2019-08-23:
Apologies for the delay. 

There are some tips in the thread here: https://github.com/ofTheo/ofxKinectV2/pull/21
I am hoping to get this merged in soon. 

Let me know if any of these tips work for this issue. 
Thanks!

amozeng, 2019-08-24:
Hey @ofTheo 
Thanks for the quite reply!

in [#21](https://github.com/ofTheo/ofxKinectV2/pull/21)
> ofxKinectV2\libs\libusb\lib\vs\libusb-1.0.dll copied into the application's bin folder
helps to fix this problem! ( libusb-1.0.dll)

However I encounter some other problem related to MSVCR120D.dll It might related to my version of VS, will keep digging into this and try to fix it.

Thanks for the info!

ofTheo, 2019-10-18:
closed by #39

closed 2019-10-18 by ofTheo