ofTheo/ofxKinectV2 #24: Don't grab RGB frame

BaptisteTheoriz, 2016-08-10:
Hi, 

I try to optimize my app performances, and I noticed that there are some processes relative to the RGB frame that I could avoid because I just use depth and IR streams.
But there isn't any simple way to do it in the addon.

The first thing I noticed is that there are some parameters in ofProtonect that are available for that, in the openKinect function, when creating a new libfreenect2::SyncMultiFrameListener object. 
The parameters are currently "libfreenect2::Frame::Color | libfreenect2::Frame::Ir | libfreenect2::Frame::Depth", which takes all streams. 
It could be nice to be able to set which stream we want to grab. 

I tried to remove the libfreenect2::Frame::Color parameter, but I don't know why, some functions which seems to be relative to the RGB stream are still called. Particularly the function ofRGBPacketProcessor::process which takes a lot of CPU usage with multiple kinects.

I have another problem which may (or not) be relative to that question : with 4 kinects on a computer (each one on a dedicated PCI USB3.0 bus), libusb gives me a "LIBUSB_ERROR_NO_MEM : Insufficient memory." error when opening the 4th kinect. I don't know if avoiding RGB could solve that problem, but if you have any suggestion, I'm willing to try anything.

Thanks !

ofTheo, 2019-08-23:
@BaptisteTheoriz yes I think the next thing to do would be to add maybe a ofxKinect::Settings object which we could pass in to open or a setup function to enable/disable the different streams of data. 

Right now I think everything is being grabbed.

ofTheo, 2019-10-18:
closed by #39

closed 2019-10-18 by ofTheo