ofTheo/ofxKinectV2 #25: The hardware requirement of multiple Kinect v2

Severn-Vergil, 2016-08-13:
How does ofxKinectV2 work with multiple Kinect V2? And what's the minimum hardware requirement for running them stably?
As I use libfreenect2 on Ubuntu 14.04 with my notebook, when I try to run Protonect with 2 Kinects, only one of them will work, the other programs will break down with errors like this:
[Error] [protocol::UsbControl] failed to set configuration! LIBUSB_ERROR_OTHER Other error. Try debugging with environment variable: export LIBUSB_DEBUG=3 .
[Info] [Freenect2DeviceImpl] closing...
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_IO Input/Output Error
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_IO Input/Output Error
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Error] [Freenect2Impl] failed to open Kinect v2: @2:6

Thank you in advance! The work is really helpful as it supports multiple Kinects!

ofTheo, 2016-08-13:
It should work if you have multiple seperate usb 3 hubs. If they show up under the same hub it won't work. On my Mac book pro when I plug them in they are listed seperately. So both can use the full super speed bandwidth. 

One other weird quirk is on OS X I have to unplug the power to them and replug it in after they are plugged in to my laptop to get them to be listed as superspeed correctly. 

I hope that helps!

Severn-Vergil, 2016-08-13:
Here is my result of lsusb:
## lsusb
Bus 004 Device 002: ID 8087:8000 Intel Corp. 
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 8087:8008 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 009: ID 045e:02d8 Microsoft Corp. 
Bus 002 Device 005: ID 045e:02d9 Microsoft Corp. 
Bus 002 Device 010: ID 045e:02d8 Microsoft Corp. 
Bus 002 Device 007: ID 045e:02d9 Microsoft Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 04f2:b40d Chicony Electronics Co., Ltd 
Bus 001 Device 007: ID 045e:02d9 Microsoft Corp. 
Bus 001 Device 003: ID 062a:4101 Creative Labs Wireless Keyboard/Mouse
Bus 001 Device 006: ID 138a:0050 Validity Sensors, Inc. Swipe Fingerprint Sensor
Bus 001 Device 008: ID 045e:02d9 Microsoft Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

##lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
    |__ Port 1: Dev 7, If 0, Class=Hub, Driver=hub/1p, 5000M
    |__ Port 3: Dev 5, If 0, Class=Hub, Driver=hub/1p, 5000M
        |__ Port 1: Dev 9, If 0, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 9, If 1, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 9, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 1: Dev 9, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/14p, 480M
    |__ Port 1: Dev 8, If 0, Class=Hub, Driver=hub/1p, 480M
    |__ Port 2: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 2: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 3: Dev 7, If 0, Class=Hub, Driver=hub/1p, 480M
    |__ Port 7: Dev 5, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 7: Dev 5, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 14: Dev 6, If 0, Class=Vendor Specific Class, Driver=, 12M

Does the result show that my notebook has only one usb 3 hub? If it is a hardware problem, maybe I have to use workstation instead. Thank you so much for your quick reply!

closed 2019-10-15 by ofTheo