ofTheo/ofxKinectV2 #27: Unable to build under Linux

aroman, 2016-11-18:
Here's my complete build log. I have been able to build successfully under macOS, but running on elementary OS/Ubuntu, I get these errors about` Logging.h`. I have the same problem building under both Clang and GCC. I'm not sure what would be different across the platforms...

```aroman@sensei:~/Developer/openFrameworks/addons/ofxKinectV2% cd example 
aroman@sensei:~/Developer/openFrameworks/addons/ofxKinectV2/example% ls
addons.make  config.make  Makefile                      openFrameworks-Info.plist  src
bin          DerivedData  ofxKinectV2Example.xcodeproj  Project.xcconfig
aroman@sensei:~/Developer/openFrameworks/addons/ofxKinectV2/example% make
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl glfw3 gl glu glew gtk+-3.0 
Compiling OF library for Release
make[1]: Entering directory '/home/aroman/Developer/openFrameworks/libs/openFrameworksCompiled/project'
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl glfw3 gl glu glew gtk+-3.0 
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl glfw3 gl glu glew gtk+-3.0 
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl glfw3 gl glu glew gtk+-3.0 
Done!
make[1]: Leaving directory '/home/aroman/Developer/openFrameworks/libs/openFrameworksCompiled/project'

Compiling example for Release
make[1]: Entering directory '/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/example'
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-1.0 gstreamer-1.0 gstreamer-video-1.0 gstreamer-base-1.0 libudev freetype2 fontconfig sndfile openal openssl glfw3 gl glu glew gtk+-3.0 
cat: obj/linux64/Release/.compiler_flags: No such file or directory
Compiling /home/aroman/Developer/openFrameworks/addons/ofxKinectV2/example/src/ofApp.cpp
g++ -c -O3 -DNDEBUG -march=native -mtune=native -Wall -std=c++14 -DGCC_HAS_REGEX -DOF_USING_GTK -DOF_USING_GTK  -pthread -I/usr/include/gstreamer-1.0 -I/usr/lib/x86_64-linux-gnu/gstreamer-1.0/include -I/usr/include/AL -I/usr/include/libdrm -I/usr/include/gtk-3.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib/x86_64-linux-gnu/dbus-1.0/include -I/usr/include/gtk-3.0 -I/usr/include/gio-unix-2.0/ -I/usr/include/mirclient -I/usr/include/mircommon -I/usr/include/mircookie -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/freetype2 -I/usr/include/libpng12 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libpng12 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/home/aroman/Developer/openFrameworks/libs/fmodex/include -I/home/aroman/Developer/openFrameworks/libs/glfw/include -I/home/aroman/Developer/openFrameworks/libs/glfw/include/GLFW -I/home/aroman/Developer/openFrameworks/libs/kiss/include -I/home/aroman/Developer/openFrameworks/libs/poco/include -I/home/aroman/Developer/openFrameworks/libs/tess2/include -I/home/aroman/Developer/openFrameworks/libs/utf8cpp/include -I/home/aroman/Developer/openFrameworks/libs/utf8cpp/include/utf8 -I/home/aroman/Developer/openFrameworks/libs/openFrameworks -I/home/aroman/Developer/openFrameworks/libs/openFrameworks/sound -I/home/aroman/Developer/openFrameworks/libs/openFrameworks/video -I/home/aroman/Developer/openFrameworks/libs/openFrameworks/app -I/home/aroman/Developer/openFrameworks/libs/openFrameworks/events -I/home/aroman/Developer/openFrameworks/libs/openFrameworks/math -I/home/aroman/Developer/openFrameworks/libs/openFrameworks/3d -I/home/aroman/Developer/openFrameworks/libs/openFrameworks/graphics -I/home/aroman/Developer/openFrameworks/libs/openFrameworks/utils -I/home/aroman/Developer/openFrameworks/libs/openFrameworks/communication -I/home/aroman/Developer/openFrameworks/libs/openFrameworks/gl -I/home/aroman/Developer/openFrameworks/libs/openFrameworks/types -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/example/ofxKinectV2Example.xcodeproj/project.xcworkspace -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/example/ofxKinectV2Example.xcodeproj/project.xcworkspace/xcuserdata -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/example/ofxKinectV2Example.xcodeproj/project.xcworkspace/xcuserdata/theo.xcuserdatad -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/example/ofxKinectV2Example.xcodeproj/xcshareddata -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/example/ofxKinectV2Example.xcodeproj/xcshareddata/xcschemes -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/example/src -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/example/DerivedData -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/example/DerivedData/ofxKinectV2Example -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/example/DerivedData/ofxKinectV2Example/TextIndex -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/src -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libfreenect2 -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libfreenect2/src -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libfreenect2/src/shader -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libfreenect2/include -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libfreenect2/include/libfreenect2 -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libfreenect2/include/internal -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libfreenect2/include/internal/CL -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libfreenect2/include/internal/libfreenect2 -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libfreenect2/include/internal/libfreenect2/protocol -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libfreenect2/include/internal/libfreenect2/usb -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/protonect -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libusb -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libusb/lib -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libusb/lib/vs -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libusb/lib/vs/x64 -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libusb/lib/vs/x64/Debug -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libusb/lib/vs/x64/Debug/dll -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libusb/lib/vs/x64/Release -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libusb/lib/vs/x64/Release/dll -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libusb/lib/vs/Win32 -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libusb/lib/vs/Win32/Debug -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libusb/lib/vs/Win32/Debug/dll -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libusb/lib/vs/Win32/Release -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libusb/lib/vs/Win32/Release/dll -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libusb/lib/osx -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libusb/include -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libusb/include/libusb -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libusb/include/libusb/os -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libfreenect2/include -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libfreenect2/include/libfreenect2 -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libfreenect2/include/internal -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libfreenect2/include/internal/CL -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libfreenect2/include/internal/libfreenect2 -I/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/libs/libfreenect2/ [... cut ...]
/home/aroman/Developer/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:200: recipe for target 'obj/linux64/Release/src/ofApp.o' failed
make[1]: *** [obj/linux64/Release/src/ofApp.o] Error 1
make[1]: Leaving directory '/home/aroman/Developer/openFrameworks/addons/ofxKinectV2/example'
/home/aroman/Developer/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:125: recipe for target 'Release' failed
make: *** [Release] Error 2
aroman@sensei:~/Developer/openFrameworks/addons/ofxKinectV2/example%
```

ShadowMitia, 2017-02-12:
I added `#undef  None` after `#define LIBFREENECT2_LOGGER_H_` because it seems to be conflicting with another `enum` in X11.

But after that I get "undefined reference" issues with the libs... I'm guessing that "addons_config.mk" needs to be changed to work under linux.

ShadowMitia, 2017-02-12:
I added this to the addon_config.mk

`linux64:
	ADDON_PKG_CONFIG_LIBRARIES = libusb-1.0
	ADDON_LDFLAGS = -Llibfreenect2
	ADDON_LDFLAGS += -lOpenCL`

Hopefully it'll help someone^^

closed 2019-10-15 by ofTheo