ofTheo/ofxKinectV2 #30: Kinect v2 as webcam in linux ?

Abduoit, 2017-07-18:
Can I use kinect v2 instead of webcam ?

mattfelsen, 2017-07-24:
Yes! You can get the color RGB data with `kinect.getRgbPixels()` which returns an `ofPixels` object. If you need to draw it to screen, you can upload this data to an `ofTexure` and draw that.  You can see it in the example app here:
https://github.com/ofTheo/ofxKinectV2/blob/afde06f7453a357b51b5815a7866d2b33cfd3efd/example/src/ofApp.cpp#L46

mattfelsen, 2017-07-24:
And of course after posting the comment, I noticed the part about linux. libfreenect2, which this addon uses, does indeed support linux but I'm not sure if this addon itself will work. Sorry for the somewhat misleading info!

ofTheo, 2019-10-15:
This should work fine in linux now thanks to some of the recent PRs from others.

closed 2019-10-15 by ofTheo