ofTheo/ofxKinectV2 #32: fatal error: 'ofxKinectV2.h' file not found

JiahuiZhou, 2017-10-16:
Just pull down and correct the path to the OF root path, then make, get this error. 
Is there anyone who know how can I fix this?
----

HOST_OS=Darwin
Compiling OF library for Release
HOST_OS=Darwin
HOST_OS=Darwin
HOST_OS=Darwin
Done!

Compiling example for Release
HOST_OS=Darwin
Compiling /Users/zhoujiahui/Kinect/ofxKinectV2-master/example/src/main.cpp
c++ -c -Os -DNDEBUG -mtune=native -stdlib=libc++ -Wall -fexceptions -fpascal-strings -mmacosx-version-min=10.7 -D__MACOSX_CORE__ -x objective-c++ -std=c++11 -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/FreeImage/include -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/boost/include -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/boost/include/boost -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/cairo/include -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/cairo/include/cairo -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/fmodex/include -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/freetype/include -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/freetype/include/freetype2 -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/freetype/include/freetype2/config -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/freetype/include/freetype2/internal -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/freetype/include/freetype2/internal/services -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/glew/include -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/glew/include/GL -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/glfw/include -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/glfw/include/GLFW -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/glut/include -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/openssl/include -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/openssl/include/openssl -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/poco/include -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/rtAudio/include -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/tess2/include -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/utf8cpp/include -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/utf8cpp/include/utf8 -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/openFrameworks -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/openFrameworks/3d -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/openFrameworks/app -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/openFrameworks/communication -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/openFrameworks/events -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/openFrameworks/gl -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/openFrameworks/graphics -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/openFrameworks/math -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/openFrameworks/sound -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/openFrameworks/types -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/openFrameworks/utils -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/libs/openFrameworks/video -I/Users/zhoujiahui/Kinect/ofxKinectV2-master/example/DerivedData -I/Users/zhoujiahui/Kinect/ofxKinectV2-master/example/DerivedData/ofxKinectV2Example -I/Users/zhoujiahui/Kinect/ofxKinectV2-master/example/DerivedData/ofxKinectV2Example/TextIndex -I/Users/zhoujiahui/Kinect/ofxKinectV2-master/example/ofxKinectV2Example.xcodeproj/project.xcworkspace -I/Users/zhoujiahui/Kinect/ofxKinectV2-master/example/ofxKinectV2Example.xcodeproj/project.xcworkspace/xcuserdata -I/Users/zhoujiahui/Kinect/ofxKinectV2-master/example/ofxKinectV2Example.xcodeproj/project.xcworkspace/xcuserdata/theo.xcuserdatad -I/Users/zhoujiahui/Kinect/ofxKinectV2-master/example/ofxKinectV2Example.xcodeproj/project.xcworkspace/xcuserdata/zhoujiahui.xcuserdatad -I/Users/zhoujiahui/Kinect/ofxKinectV2-master/example/ofxKinectV2Example.xcodeproj/xcshareddata -I/Users/zhoujiahui/Kinect/ofxKinectV2-master/example/ofxKinectV2Example.xcodeproj/xcshareddata/xcschemes -I/Users/zhoujiahui/Kinect/ofxKinectV2-master/example/src -I/Users/zhoujiahui/Kinect/of_v0.9.8_osx_release/addons/ofxGui/src -MMD -MP -MF obj/osx/Release/src/main.d -MT obj/osx/Release/src/main.o -o obj/osx/Release/src/main.o -c /Users/zhoujiahui/Kinect/ofxKinectV2-master/example/src/main.cpp
In file included from /Users/zhoujiahui/Kinect/ofxKinectV2-master/example/src/main.cpp:2:
/Users/zhoujiahui/Kinect/ofxKinectV2-master/example/src/ofApp.h:4:10: fatal error: 'ofxKinectV2.h' file not found
#include "ofxKinectV2.h"
         ^~~~~~~~~~~~~~~
1 error generated.
make[1]: *** [obj/osx/Release/src/main.o] Error 1
make: *** [Release] Error 2

ofTheo, 2017-10-16:
Hi! 

Try using the ProjectGenerator that comes with the latest OF download to add the addon to your project or generate project files for the example.

JiahuiZhou, 2017-10-28:
Thank you! I figure it out by putting it in the addon directory.

closed 2017-10-28 by JiahuiZhou