ofTheo/ofxKinectV2 #41: Confirm linux64 and linuxarmv7l

ofTheo, 2019-10-18:
@madelinegannon @bakercp 

@NickHardeman and I have done a cleanup to ofxKinectV2 - going back to using the libfreenect source to make it easier to build for mac and windows. 

Both mac and windows build fine now with the projectGenerator.
Could you check on Linux and see if anything needs adding to addons_config.mk? 

Lots of nice new features like ofxKinecV2::Settings which allows for streams to be toggled on and off and max depth range to be changed. 

See: https://github.com/ofTheo/ofxKinectV2/blob/master/src/ofxKinectV2.h#L25

edap, 2020-03-02:
Hi. I have followed the amazing tutorial by @madelinegannon https://gist.github.com/madelinegannon/10f62caba7184b90ea43a734768e5147 and I was able to run the `./bin/Protonect` executable.

I have installed OF and compiled with gcc-6. I have added these lines to my .bashrc file (and I hame made sure that cuda was correctly installed and that `/usr/local/cuda` is the correct location).

```
export LD_LIBRARY_PATH="/usr/local/cuda/lib64:${LD_LIBRARY_PATH}"
export LD_LIBRARY_PATH="/usr/local/lib:${LD_LIBRARY_PATH}"
export PATH="/usr/local/cuda/bin:${PATH}"
```

When I run the example provided in this repository, I receive the error: "of_v20200227_linux64gcc6_release/addons/ofxKinectV2/libs/protonect/ofProtonect.h:8:41: fatal error: libfreenect2/libfreenect2.hpp: No such file or directory
 #include <libfreenect2/libfreenect2.hpp>
"

and actaully there is no `libfreenect2.so` in `/usr/local/lib`.

Any clue? I think that libfreenect is probably not installed system wide, so the Proton example works, but the OF app no.

edap, 2020-03-04:
Just as a side note, I had make a mistake while installing libfreenect2. The instruction reported by @madelinegannon are correct, just:

```
> mkdir build && cd build
> cmake -DBUILD_SHARED_LIBS=ON ..
> make
> make install
```

needs to be 

```
> mkdir build && cd build
> cmake -DBUILD_SHARED_LIBS=ON ..
> make
> **sudo** make install
```
Otherwise it can not install the library system wide.
The example application provided in  works in the .
I have copied that example found in `https://github.com/madelinegannon/ofxKinectV2` in this repository.

I have removed all the libraries in `addons/ofxKinectV2/libs` except of `protonect`.
I have run `make && makeRunRelease` and the example worked.

ofTheo, 2020-03-04:
Thanks Davide!

We can take a look and see what's in Madeline's example that we can add or do a example-linux as suggested.

edap, 2020-03-04:
The only two steps to get the example working on linux (after installing cuda, libfreenect2, etc, as explained in the guide) are:

-  Add `PROJECT_LDFLAGS=-L/usr/local/lib/ -lfreenect2 -L/usr/local/cuda/lib64 -lcuda -lcudart` to config.make
- Delete everything except `protonect` in `ofxKinectV2/libs`

Then if you run `make && make RunRelease` it works

ofTheo, 2020-03-04:
Awesome! Thank you!