ofTheo/ofxKinectV2 #42: Can't load .jpg images when using ofxKinectV2

kaua-melo, 2019-11-04:
Thank you very much for the addon, Theo! 

I came across the following issue. I can't open a .jpg image if I have ofxKinectV2 in the _addons.make_ file. Loading .png works fine.

_addons.make_
```
ofxKinectV2 <- Remove this to be able to load .jpg
```

_ofApp.h_
```
#include "ofxKinectV2.h" //  comment this to be able to load .jpg
...
    ofImage bImg;
```

_ofApp.cpp_
```
...
void ofApp::draw()
{
    if (bImg.isAllocated())
    {
        bImg.draw(0, 0);
    }
}
...
void ofApp::dragEvent(ofDragInfo dragInfo)
{
    bImg.load(dragInfo.files[dragInfo.files.size() - 1]);
    cout << "Path: " << dragInfo.files[dragInfo.files.size() - 1] << endl;
    cout << "Dimensions: " << bImg.getWidth() << "x" << bImg.getHeight() << endl;
}
```

When I run the code above and drag a .jpg onto the canvas, I get the following on the terminal:

```
[ error ] ofImage: loadImage(): couldn't load image from ""/Users/kaua/Desktop/asd.jpg""
Path: /Users/kaua/Desktop/asd.jpg
Dimensions: 0x0
```

I'm using of_v0.10.1_osx_release on macOS 10.14.4.

ofTheo, 2019-11-04:
Thanks for this report. Will test this out. 
I wonder if this is due to the addition of libturbojpeg conflicting with freeimage in some way.

NickHardeman, 2019-11-04:
Just tested this on OSX with an app built with the project generator and was able to load both jpg and png. As Theo mentioned, libturbojpeg could be interfering if it's added to Xcode or the lib is linked. That lib is not needed on OSX. If it's added try removing it and see if that fixes it.

ofTheo, 2019-11-04:
Hmm weird - I am on macOS 10.14.6 and also can't recreate this issue. 
I even tried dragging both libturbojpeg libs into the project.  

Are you using Xcode or another environment / make files?

kaua-melo, 2019-11-05:
Hey, thanks for the super fast response.

That's weird, I'm not sure what I'm missing here.
Here's what I just tried: 
- downloaded OF of_v0.10.1_osx_release and ofxKinectV2 again.
- pasted ofxKinectV2 in the addons folder.
- duplicated the emptyExample and renamed it.
- Pasted the code I described above.
- From the Visual Studio Code terminal:
```
$ make
$ make RunRelease
```

As Nick suggested, I also tried to remove libturbojpeg. I only found mentions to it under _addons/ofxKinectV2/addon_config.mk_, so I commented these lines:
```
    #ADDON_INCLUDES_EXCLUDE  = "libs/libturbojpeg"
    #ADDON_INCLUDES_EXCLUDE  += "libs/libturbojpeg/%"
```
but I still have the same issue.

When I check the _obj/osx/Release/.compiler_flags_ file after compiling the code, I can see the following mentions to libturbojpeg:

```
-I/Users/kaua/Documents/of_v0.10.1_2/addons/ofxKinectV2/libs/libturbojpeg
-I/Users/kaua/Documents/of_v0.10.1_2/addons/ofxKinectV2/libs/libturbojpeg/include
-I/Users/kaua/Documents/of_v0.10.1_2/addons/ofxKinectV2/libs/libturbojpeg/lib
-I/Users/kaua/Documents/of_v0.10.1_2/addons/ofxKinectV2/libs/libturbojpeg/lib/vs
-I/Users/kaua/Documents/of_v0.10.1_2/addons/ofxKinectV2/libs/libturbojpeg/lib/vs/x64
-I/Users/kaua/Documents/of_v0.10.1_2/addons/ofxKinectV2/libs/libturbojpeg/lib/vs/x64/Release
-I/Users/kaua/Documents/of_v0.10.1_2/addons/ofxKinectV2/libs/libturbojpeg/lib/vs/x64/Debug
-I/Users/kaua/Documents/of_v0.10.1_2/addons/ofxKinectV2/libs/libturbojpeg/lib/vs/Win32
-I/Users/kaua/Documents/of_v0.10.1_2/addons/ofxKinectV2/libs/libturbojpeg/lib/vs/Win32/Release
-I/Users/kaua/Documents/of_v0.10.1_2/addons/ofxKinectV2/libs/libturbojpeg/lib/vs/Win32/Debug
-I/Users/kaua/Documents/of_v0.10.1_2/addons/ofxKinectV2/libs/libturbojpeg/lib/osx
-I/Users/kaua/Documents/of_v0.10.1_2/addons/ofxKinectV2/libs/libturbojpeg/lib/linuxarmv6l
-I/Users/kaua/Documents/of_v0.10.1_2/addons/ofxKinectV2/libs/libturbojpeg/lib/ios
```

When I remove the ofxKinectV2 from the _addons.make_ file, there are no mentions to libturbojpeg in _obj/osx/Release/.compiler_flags_ and the code works fine. 

So apparently ofxKinectV2 is still making usage of libturbojpeg? Do you have any suggestion on how can we avoid that?

ofTheo, 2019-11-05:
So looks like a vscode / makefile quirk as Xcode doesn't seem to have these issues. 

Either way the ADDON_INCLUDES_EXCLUDE should be excluding the files. 

That might be the project generator not picking up the includes correctly. 
This might be fixed in the nightly build of OF here: https://openframeworks.cc/download/ 

If you delete this folder all together from Finder does it build and run okay?
`addons/ofxKinectV2/libs/libturbojpeg/lib`

kaua-melo, 2019-11-05:
Good news! 

Deleting the folder `addons/ofxKinectV2/libs/libturbojpeg/lib` seems to solve the problem. 
It loads .jpg now with no errors.

I think the issue might be with makefile since I get the same issue when I'm using the Terminal.

Thank you very much Theo, I appreciate the fast responses! You work is really inspiring.