ofTheo/ofxKinectV2 #44: How to make it run in VS 2019? - getting LIBUSB_ERROR_NOT_SUPPORTED

maximoskp, 2019-12-02:
Hi - thanks again for the Addon!

Trying to run it in VS 2019 on Windows 10 and after the application window has launched properly but with black screens (no image), in the app's console I get four errors related to LIBUSB_ERROR_NOT_SUPPORTED:

- [Error] [Freenect2Impl] failed to open Kinect v2: @1:9 LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform (TWICE)

- [Error] [protocol::UsbControl] failed to claim interface with IrInterfaceId(=1)! LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform. Try debugging with environment variable: export LIBUSB_DEBUG=3 .

- [Error] [Freenect2Impl] failed to open Kinect v2: @1:9

Also the device does not seem to be properly activated, i.e. I get no red light triplet on the front panel.

Any idea how I can make it work on VS 2019?
Thanks!

NickHardeman, 2019-12-02:
Did you try updating the driver on the kinect using zadig as described in the Read Me?

maximoskp, 2019-12-03:
Thanks for your response! I'm getting this error while having already updated the driver with zagid. It appears that the update has been successful, at least as far as I can figure out by the "Reinstall driver" rather than "Replace driver" text on the button when selecting the libusbK (v3.0.7.0) driver.

[image]

NickHardeman, 2019-12-04:
Hmm, that seems correct. I had trouble with some usb ports, I had to try several different USB3 ports before it registered properly. Have you restarted and tried unplugging and then plugging in the kinect? I just downloaded the latest release of OF from the site and this addon, built the kinectV2 example with the project generator and it worked for me. Maybe you need to use Visual Studio 2017?

eulphean, 2019-12-04:
Unfortunately, I am seeing the same issue with Visual Studio Community 2017 as @maximoskp. Window launches but all the screens are black (no image) 
I'm running Windows 10 ona dual-boot macbook pro with Bootcamp. @NickHardeman, you think this could have something to do with? KinectV2 runs fine on OSX.

eulphean, 2019-12-04:
@maximoskp - Make sure you uncheck Ignore Hubs or Composite Parents. 
Then select XBOX Nui Sensor from the list to install libusbK driver. 

[image]

maximoskp, 2019-12-06:
Thank you @eulphean - this did the trick for me! No error messages now and all screens show as expected in the demo. Thanks a million :)

closed 2019-12-06 by maximoskp