ofTheo/ofxKinectV2 #45: Windows error after succesfull build

ReinierGielen, 2020-01-29:
Hello, I am trying to get your addon working.
I have a WIN10 installation with Visual Studio 2017.
I installed the nightly build of openframeworks and had several successfull projects.

I added ofxKinectV2 but I can't figure out why I am getting an error on startup.
The build gives no error. But on start I am getting:  The program '[53172] kinect_debug.exe' has exited with code -1073741701 (0xc000007b).
[image]

My properties are set like this:
{
  "configurations": [
    {
      "inheritEnvironments": [
        "msvc_x86"
      ],
      "name": "x86-Debug",
      "includePath": [
        "${env.INCLUDE}",
        "${workspaceRoot}\\**"
      ],
      "defines": [
        "WIN32",
        "_DEBUG",
        "UNICODE",
        "_UNICODE"
      ],
      "intelliSenseMode": "windows-msvc-x86"
    }
  ]
}

Kan you please help to get this working.

ofTheo, 2020-01-29:
Can you confirm you have installed the libusbk drivers to the composite hub? 
See: https://github.com/ofTheo/ofxKinectV2#windows-10-usb-setup

ReinierGielen, 2020-01-29:
Yes, the libusbk was setup with that manual

ofTheo, 2020-01-29:
you might need to copy the dlls from the ofxKinectV2/libs/libusb/lib/vs folder to you bin/ folder.

ReinierGielen, 2020-01-30:
Thanks for the tip, I copied the dll's from C:\of_v20200117_vs2017_release\addons\ofxKinectV2-master\libs\libusb\lib\vs\Win32\Release  to the the bin in the project folder and the error disappeared and the program window started.

No image yet as it seems we still have an usb issue:
[Error] [protocol::UsbControl] failed to claim interface with IrInterfaceId(=1)! LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform. Try debugging with environment variable: export LIBUSB_DEBUG=3 .
[Error] [Freenect2Impl] failed to open Kinect v2: @3:14

We did the Zadig install procedure, so I am not sure where this is coming from

yty, 2021-03-16:
me too.  Has anyone solved it? 

 of_v0.10.1_vs2017_release + win10.1 64

```
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 19 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:27 with serial 500322242742
[Info] [Freenect2Impl] found 1 devices
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: GeForce GTX 980  (GPU)[NVIDIA Corporation ]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: GeForce GTX 980  (GPU)[NVIDIA Corporation ]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 19 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:27 with serial 500322242742
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Error] [protocol::UsbControl] failed to claim interface with IrInterfaceId(=1)! LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform. Try debugging with environment variable: export LIBUSB_DEBUG=3 .
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Error] [Freenect2Impl] failed to open Kinect v2: @2:27
```