ofTheo/ofxKinectV2 #46: Error on Example

tobyshooters, 2020-02-26:
Hey, I'm able to compile the example and it even detects my Kinect.
However, it fails to get any input. Do I need to install any other software besides the addon?

```
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 1 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @20:13 with serial 115837341047
[Info] [Freenect2Impl] found 1 devices
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) Core(TM) i5-5257U CPU @ 2.70GHz (CPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl]   1: Intel(R) Iris(TM) Graphics 6100 (GPU)[Intel Inc.]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: Intel(R) Iris(TM) Graphics 6100 (GPU)[Intel Inc.]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 1 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @20:13 with serial 115837341047
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 4*128*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 1073741824
[Debug] [Freenect2DeviceImpl] status 0x090000: timeout
[Debug] [Freenect2DeviceImpl] status 0x090000: 1073741824
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[verbose] ofProtonect::openKinect: device serial: 115837341047
[verbose] ofProtonect::openKinect: device firmware: 4.0.3916.0
[ error ] ofProtonect::updateKinect: Timeout serial: 115837341047
[ error ] ofGLUtils: ofGetGLInternalFormatFromPixelFormat(): unknown OF pixel format OF_PIXELS_UNKNOWN, returning GL_RGBA
[ error ] ofGLUtils: ofGetGLFormatFromPixelFormat(): unknown OF pixel format OF_PIXELS_UNKNOWN, returning GL_LUMINANCE
[ error ] ofTexture: allocate(): ofTextureData has 0 width and/or height: 0x0
```

NickHardeman, 2020-02-26:
If you are on OSX, you shouldn't need anything else. On Windows you will need to update the libusbK drivers using Zadig as described in the ReadMe.

tobyshooters, 2020-03-02:
Hey, the Kinect is definitely connecting. But I'm getting a ofProtonect timeout.
I'm on a Early 2015 Macbook Pro.

[image]

```
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 1 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @20:7 with serial 115837341047
[Info] [Freenect2Impl] found 1 devices
[ error ] ofProtonect::updateKinect: Timeout serial: 115837341047
0[ error ] ofGLUtils: ofGetGLInternalFormatFromPixelFormat(): unknown OF pixel format OF_PIXELS_UNKNOWN, returning GL_RGBA
[ error ] ofGLUtils: ofGetGLFormatFromPixelFormat(): unknown OF pixel format OF_PIXELS_UNKNOWN, returning GL_LUMINANCE
[ error ] ofTexture: allocate(): ofTextureData has 0 width and/or height: 0x0
```

jessepak, 2021-11-29:
hey @tobyshooters I got the same issue as well, an update on this?

jessepak, 2021-11-30:
I got things fixed, turns out for macos you need to use a factory cable instead of the replacement cable while the later one is working fine on both window and linux system